supermileage/telemetry-firmware #7: Add a GitHub Action workflow to build firmware.

lcfyi, 2020-11-28:
As title. There are two ways to approach this: use an existing Docker configuration for the Particle CLI to compile, or install the toolchain manually and let it build against the latest build every push.

The former is probably easier.

silviu-toderita, 2020-12-05:
Created Docker container and tested locally for building Particle firmware. Pushed to new Docker organization "ubcsupermileage". Will share with Leslie once I get his username/email. Have not started the process of integrating Github Action with Docker, will likely need Leslie's guidance for that.

lcfyi, 2021-01-11:
> Created Docker container and tested locally for building Particle firmware. Pushed to new Docker organization "ubcsupermileage". Will share with Leslie once I get his username/email. Have not started the process of integrating Github Action with Docker, will likely need Leslie's guidance for that.

Could you share your `Dockerfile` here? We can skip the Docker Hub and use the packages feature Github offers to organizations ([these](https://github.com/orgs/supermileage/packages)). We can build and tag it then use that in the workflow going forward (and in our local workflows) so that we're all on the same page.

lcfyi, 2021-01-14:
83412352b71fbe84633371ff6932bd2e613b27f4 should set us on the right path with regards to setting up the build pipeline. At the moment, it targets all branches to build for the Electron since most of our code still targets that, but we can refactor the workflow to target specific branches and actions later on.

I'd say this is still a WIP but we're making progress.

lcfyi, 2021-01-20:
Throwing a note here for the remaining `TODOs` to this task:

- [ ] Add workflow that verifies pull requests too
- [ ] Convert this repo into a monorepo for firmware targeting different vehicles
- [ ] Add a conditional workflow that builds for specific vehicles

For the second task, it'd be easier if we extract the libraries out into a common directory, which will be symlinked into each directory. However, this is problematic if any of you guys are developing on Windows. Can I get a quick check as to which platform you guys are predominantly using when developing? @silviu-toderita @DFriend01 

Another option is a custom makefile that targets another directory for the libraries, so we preprocess that. However, I think this breaks the ability to build within Particle Workbench (I haven't verified whether it respects user makefiles yet), and it'd also break the symbols lookup for the code suggestions in VSCode. It's a nasty tradeoff that involves a bit of setup to get it working nicely.

Do any of you guys have any strong opinions on how we should proceed? This will affect your dev workflow when working on the firmware so I'd like your guys' input. We can also talk more about the options over VC this Saturday if you'd like.

PR #14 (2021-01-24): Split code into libraries and vehicle-dependent firmware.
Closes #7. 

There are caveats about this in Windows, so please check out the README when cloning.
closed 2021-01-24 by lcfyi