supermileage/telemetry-firmware #8: Consolidate existing ECOTRONS and CAN libraries into this repo.

lcfyi, 2020-11-28:
Instead of using submodules, we'll just write the code directly here.

[ECOTRONS](https://github.com/lcfyi/ECOTRONS-RS232-telemetry)

[CAN Telemetry](http://github.com/lcfyi/can-telemetry)

silviu-toderita, 2020-11-28:
Opened pull request for ECOTRONS integration and copied over CAN Telemetry library source, will integrate and test CAN Telemetry next weekend.

silviu-toderita, 2020-12-05:
CAN Telemetry will be hard to integrate and test without a CAN device to test with. I will order a couple more CAN interfaces on Aliexpress, but it's a toss-up if they will arrive before we get access to Rusty.

silviu-toderita, 2020-12-05:
Next time we meet, we should talk about how to integrate GPS and the ECU communication. I noticed in past repos, they're never running at the same time. Boron only has 1 serial port, but we can likely use a software serial library for the GPS module since it's running at a fairly low speed (9600).

lcfyi, 2021-01-11:
> Next time we meet, we should talk about how to integrate GPS and the ECU communication. I noticed in past repos, they're never running at the same time. Boron only has 1 serial port, but we can likely use a software serial library for the GPS module since it's running at a fairly low speed (9600).

This is a good point, we didn't have issues with this on the Electron since there were multiple UART interfaces, but the Boron definitely has this problem.

I know there are breakout boards that we can use to talk to the UART interface over SPI or I2C, but that means we'll have to account for them in our PCB design as well (which I'd like to talk about this weekend as we should probably have those finalized by the end of the month). At the same time, location might not be too necessary if we decide to drop it since it has always been a pain point with regards to lock times; we aren't allowed to keep the GPS module active with an external battery, which means it needs to cold start every time the vehicles are power-cycled.

> CAN Telemetry will be hard to integrate and test without a CAN device to test with. I will order a couple more CAN interfaces on Aliexpress, but it's a toss-up if they will arrive before we get access to Rusty.

Were you able to investigate this further over the break? No worries if you haven't, just curious.

silviu-toderita, 2021-01-11:
Yes, I received a bunch more MCP2515 CAN boards from Aliexpress over the break but I have not yet used them. I have 4 in total and can give 2 to Devon for testing.

closed 2021-01-16 by lcfyi