supermileage/telemetry-firmware #83: Improve null-terminated array of commands in vehicle source files

silviu-toderita, 2022-02-06:
In the following files: 

- vehicleFc.cpp
- vehicleUrban.cpp
- vehicleProto.cpp

We create a null-terminated array of all commands and then pass it to the LoggingDispatcherBuilder to built the dispatcher with all of our commands. This is clunky because all commands have to be added to this array, and we always have to remember to terminate the array with NULL. At a minimum, convert this into a vector, but ideally make it so we don't have to construct a list in addition to the LoggingCommands themselves.

cosparks, 2022-02-07:
I've already dealt with this issue on the current branch I've been working on, so I'll go and close this now

closed 2022-02-07 by cosparks

silviu-toderita, 2022-02-07:
Oh that's great! I'm not sure if we discussed this, but @valentino-jaber and I went through all the commands yesterday and updated them so that they match the sensor spreadsheet. We also changed all the remaining integer getters to return ints instead of String. That work is in the branch develop_misc

cosparks, 2022-02-07:
I saw the work you guys did, nice job!  I realized last week how cumbersome those static pointer arrays are to maintain when adding/removing logging commands so decided something needed to be done.  I just merged develop_misc onto that branch and should have a pr up tomorrow at some point