supermileage/telemetry-firmware #152: Imrpove accelerometer calculations

cosparks, 2023-08-09:
Currently acceleration is calculated using a weighted exponential moving average.  Still, the readings are quite inaccurate and could be significantly improved.

We are only able to log data (acceleration included) once per second, so some more work needs to go into finding and implementing an algorithm which computes a reasonable weighted average of the acceleration which occurs over that time period, and which also attempts to filter out some of the random noise from vibration.

cosparks, 2024-01-17:
To add to this, the other major factor in the quality of accelerometer readings is the method by which I attempted to filter gravity from the signal. The firmware attempts to use gyroscope readings to compute the x-rotation of the vehicle while it's in motion, then uses the rotation to compute how much gravity should be subtracted from y and z acceleration.  This worked quite well when the telemetry module was sitting on a desk, but clearly fails when the vehicle is running, with all the added vibration and randomness.

That said, I think the work for this issue should be something like:
1. Figure out a more reliable way to filter gravity
2. Consider how the accelerometer data could be more accurately averaged over 1s intervals*
 
*it might be worth investigating to see if Gaussian smoothing is computationally feasible. Otherwise I think using a moving uniform or exponential average is probably fine. The best way to test this would be to get some real accelerometer data sampled at 104Hz (this is the sample rate of our accelerometer), load it into a python jupyter notebook session and experiment with different filters (1d uniform, exponential and gaussian filters available in `scipy.ndimage` library). Once you have an idea of what type of filter you want and (standard deviation for Gaussian smoothing, alpha for exponential mean, etc..), then you could implement it in the firmware. You will definitely need to consider latency depending on which filter you use (filtering live data using a moving uniform average with a window of 1s adds 500ms latency to your readings, for example).

KevinZhiyuanDing, 2024-10-09:
Accelerometer calculations may be used to calculate the distance travelled by the vehicles. Must make sure that method can suppress noise and also be used in calculating average values.